Salandora/OctoPrint-FileManager #1: FileManager tab not available

ignaworn, 2015-12-13:
As discussed here https://github.com/foosel/OctoPrint/issues/1142 FileManager Tab is not appearing in the main page. I've tried disabling all the other plugins and did not work either.

Plugin Version: FileManager (0.1.0)
OctoPrint Version: 1.3.0.dev761+g28eb436 (devel branch)
Browser: Google Chrome Version 47.0.2526.80 m

Octoprint.log http://pastebin.com/gzEMi3cn

Screenshots:
[image]
[image]

Thank you!

Salandora, 2015-12-13:
That's very very interesting...
I can see that Octoprint is loading the FileManager  plugin correctly.
At least it looks like.
I gonna try to reproduce this.
Might be a bug with the version check I implemented....
If you like to test you can comment 
https://github.com/Salandora/OctoPrint-FileManager/blob/master/octoprint_filemanager/__init__.py#L402-L404
this lines in you local install ( /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filemanager)

ignaworn, 2015-12-13:
Excellent, I've added a # in each line you mentioned, restarted OctoPrint and voila!
Thank you very much!

Screenshot
[image]

Salandora, 2015-12-13:
Thanks for spotting and testing this. Guess I have to recheck the compatibility check then. :-/ very strange for me it worked when I tested it... hmm

Salandora, 2015-12-13:
I found the bug, it occured only with a version of setuptools wich gives back a tuple instead of an object.
I fixed it, updated the Version to 0.1.1 and deleted 0.1.0.
Please report back if it now works for you.

ignaworn, 2015-12-13:
Tested the latest commit https://github.com/Salandora/OctoPrint-FileManager/commit/5ceddcac68f9a3bf2a085bcfb07d9d735bd73128 and worked flawlessly. 

I think this issue can be closed. Thank you for helping

Salandora, 2015-12-13:
Lol this was a quick anser. Closing this then ;-)

closed 2015-12-13 by Salandora