Salandora/OctoPrint-FileManager #2: Creating a folder in a folder, creates a folder in root

kphill42, 2016-11-02:
When clicking into a folder and creating a sub-folder, the new "sub-folder" appears in the root folder rather than as a sub-folder of the current folder.

Current workaround is to cut and paste the folder from the root into the desired folder.

Salandora, 2016-12-10:
Sorry, forgot to reply on this: well I'm aware of this bug, but didn't had the chance to look into it.
Going to do this this weekend.

Salandora, 2016-12-14:
Well okay, got the problem, still thinking about how to solve it without code duplication...

Problem is:
I'm reusing the createFolder dialog of OctoPrint, this one is using a function inside the filesviewmodel which refers to the currentPath observable and exactly this is the problem.
Looks like I have to copy the dialog back in and make my own function again which of course is not so nice :-/

Salandora, 2017-03-08:
Should be fixed now.

closed 2017-03-08 by Salandora