Salandora/OctoPrint-FileManager #4: Copy/Cut and paste does not work

derpicknicker1, 2016-12-23:
When i try to copy/cut and paste, i get this errors in JavaScript-Console:

**For copy/cut**
```
 OctoPrint.files.pathForElement has been renamed to OctoPrint.files.pathForEntry, please use that instead
pathForElement @ packed_client.js?dc5a235f:773
(anonymous) @ packed_plugins.js?946fd97d:6018
Pn @ packed_libs.js?8f75e9e5:10655
(anonymous) @ packed_libs.js?8f75e9e5:10676
FilemanagerViewModel.self.copy @ packed_plugins.js?946fd97d:6017
(anonymous) @ packed_libs.js?8f75e9e5:10840
dispatch @ packed_libs.js?8f75e9e5:4737
elemData.handle @ packed_libs.js?8f75e9e5:4549
```

**For paste**
```
POST http://octopi.local/api/files/local/sat_flightC.gcode 409 (CONFLICT)
send	@	packed_libs.js?8f75e9e5:9175
ajax	@	packed_libs.js?8f75e9e5:8656
OctoPrint.ajax	@	packed_client.js?dc5a235f:95
OctoPrint.ajaxWithData	@	packed_client.js?dc5a235f:104
OctoPrint.post	@	packed_client.js?dc5a235f:116
OctoPrint.postJson	@	packed_client.js?dc5a235f:129
OctoPrint.issueCommand	@	packed_client.js?dc5a235f:245
issueEntryCommand	@	packed_client.js?dc5a235f:623
copy	@	packed_client.js?dc5a235f:738
FilemanagerViewModel.self.paste	@	packed_plugins.js?946fd97d:6052
(anonymous)	@	packed_libs.js?8f75e9e5:10840
dispatch	@	packed_libs.js?8f75e9e5:4737
elemData.handle	@	packed_libs.js?8f75e9e5:4549
```

I'm using Octoprint 1.3.0 (master branch)

Salandora, 2016-12-23:
The warning for Copy/Cut is okay, it's just a hey the function you wanna use has been renamed, please change this.

The Error on Paste is the real problem, can you please provide more Informations on what you have done, e.g. did you printed while you wanted to copy/move files around, do you have accessControl enabled or disabled.

One more thing: if there is a file with the same name where you wanted to paste the files, well then the plugin will abort without a message at the moment, it is supposed to ask if you want to overwrite the file but this hasn't been implemented until now. 
So it then is not a bug but a missing feature ^^.

derpicknicker1, 2016-12-23:
AccessControl is enabled. I'm not printing. I restarted the raspberry to have a 'clean boot', loaded the web interface, marked some files and clicked 'cut'. Then i marked a folder and clicked 'paste'. Nothing happened. If i select only one file, i get the error. When i select multiple files, i don't get the error but nothing is happen too.
There are no files in the folders.
Rights for folders: drwxr-xr-x 
Rights for the files: -rw-------
The folders where made by your plugin/octoprint

Salandora, 2016-12-23:
Ahh you selected the folder, well that's the wrong way ^^
You'll need to be inside the folder (double click) and then paste.
Maybe it would be interesting to implement this too sou you don't need to be inside the folder.

derpicknicker1, 2016-12-23:
Yes...i just discovered this too 😅 
Sorry. I tried it this way, but it seemed not to work. Now it does like it should. I'm not sure what i did wrong on my first tries.
Thanks a lot for your help.

closed 2016-12-23 by derpicknicker1