Salandora/OctoPrint-FileManager #5: Enable upload of generic files

amedranogil, 2017-05-10:
This plugin is perfect to share slicer presets, configurations and printing manuals (having all 3D printing resoureces in the same service).
But it only allows to manage model or toolpath files.
Can it be configured to allow uploads of generic files from admins? 
Can there be some permission management (eg: admins can upload, users can only download for a particular folder)?

Salandora, 2017-05-10:
1. Well sadly no, because this plugin is only using what octoprint is offering, + a few little additions like SD upload of existing files, bulk operations... 
2. This will partially be possible with the new Granular Permission System which I'm actually developing.
There you can say that a group of users do not have the permission to upload files, but to download.
But it is not folder specific.

amedranogil, 2017-05-16:
Well, I guess the only option is to create my own plugin which enables navigation from given dir. files would be uploaded through ssh.

closed 2017-05-16 by amedranogil

Salandora, 2017-05-17:
@amedranogil one thing I can notice here:
I once wrote a plugin for someone else who had a similiar request https://github.com/Salandora/OctoPrint-StorageManager it's been a while since that and I never did something with it or used it for my own purposes but maybe it is a starting point for you, I just have to say, I have no idea if it is still working ^^

amedranogil, 2017-05-17:
I have been testing it for some time, definitelly needs some review. I'll try to spare some time on it.
Thanks!