Salandora/OctoPrint-FileManager #9: Rename a file fails to work on just-uploaded files

robertwallace, 2017-08-25:
[ I am also using the Autoselect plugin. ]

It appears (or is a coincidence) that while Octoprint is actively calculating the print time on a newly uploaded (vi Slic3r's Send to Printer), and you're using FileManager to rename the file, the rename fails (without an error message or any other indicator) and the file is not renamed.  After Octoprint posts the calculated print time, you can now go back and successfully rename the file.

Don't know if it's related, but if you rename a file and the name is already used by another file, the rename fails without error message or other indicator that the rename was not successful.

robertwallace, 2017-08-29:
... more specifically, if the screen refreshes behind the open File Manager rename dialog box, the change is ignored.

Salandora, 2017-08-30:
Sadly this is true, I never came around to implement any error messages or other indicators, it's still on the todo list, but I lack the time for it.

The f the screen refreshes behind the open File Manager rename dialog box part sounds interesting, could you give an example how to reproduce it.

robertwallace, 2017-09-20:
It can be reliably reproduced by sending a file to the printer (via Prusa3D Slic3r's and Octoprint's  API (with key).  Note, again, that I'm also using the Autoselect plugin that will, when not printing, automatically select the just-uploaded file for printing (which starts Octoprint print time calculation.)  

If you open File Manager and select a file (for renaming  or whatever) before the print time calculations are posted (refreshing the Octoprint main screen) the file becomes unselected in File Manager.  If you have the File Manager rename dialog box open at the time, your rename action is ignored.

You may also be able to reproduce it by getting Octoprint to refresh the window  (without Autoselect installed) if you select a file from the uploaded file list, and then opening and selecting a file in File Manager before Octoprint finishes its print time calculations.

Salandora, 2017-09-21:
Ahh I see, alright I can now think of why this happens.
Okay this could be a logic mistake, will take a look soon (If I can fit it into the weekend).