Salandora/OctoPrint-FileManager #11: Filemanager appears broken on latest Octoprint release 1.3.6

robertwallace, 2017-12-13:
After latest 1.3.6 update of Octoprint, Filemanager no longer lists files.

ICode4Hugs, 2017-12-14:
In my case, only the .gcode files currently in ~./octoprint/uploads are being displayed and no directories are listed.  I have a "Printed" folder ~./octoprint/uploads/Printed and that folder doesn't show up in File Manager

FileManager 0.1.2
OctoPrint 1.3.6

MStylesMS, 2017-12-15:
Same problem here.  No files or folders listed.

robertwallace, 2017-12-17:
Thanks, ICode4Hugs,  Turning "Mixed" on brought back the files.  I had really missed being able to rename the gcode files.  Back in business...

derpicknicker1, 2017-12-23:
Same here. I only get the files listetd when `Mixed` is selected.

friesandacoke, 2017-12-25:
ICode4Hugs same. Thanks for the workaround. :(

Salandora, 2017-12-27:
Thanks for reporting this, sadly I won't have the time to check this before new year :-(

ICode4Hugs, 2017-12-27:
@Salandora Have a happy, safe and prosperous new year!

3axis, 2017-12-31:
Me too, no files, mixed options helps.

Desterly, 2018-01-18:
Pull request #13  to resolve #

Salandora, 2018-03-04:
I merged @Desterly PR right now into devel, If everything went fine, and I can do some more stuff on this plugin this week then there will be a new version next weekend.

Wolfsblvt, 2018-05-05:
Still broken, sadly :-/

robertwallace, 2018-05-28:
Yes - issue appears to be fixed.  Thanks!  Does it only get a new version number when it's released?

Also, had a problem with the install.  It turned out to be my problem, but here it is in case others see the same behavior:  [this was a re-install, but I also saw it with an uninstall/install]  Even after restarting the OctoPrint server using the "Restart Now" button that pops up after installation, I still wasn't seeing the working version.  It was only after I refreshed the Octoprint web page (forcing a new login) that the working version was running.   Seems that "Restart Now" button doesn't always force a new session - probably a feature -- and if you don't get a new session, you don't get the new FileManager version.  I'm not sure if clearing the browser cache would kick-start it - I didn't try.

Anyway, my 6 OctoPrint installations now have working FileManagers.  Thanks, again, for your effort in fixing this!

robertwallace, 2018-05-28:
I had forgotten that I was the one that opened this issue.  I consider it fixed, so I'm closing it but I'm not sure what closing the issue does to the comment section.  It's not my intention to stop comments.  Let me know if that happened. [if you can] <grin>

closed 2018-05-28 by robertwallace

Wolfsblvt, 2018-05-29:
@robertwallace Anyone can still comment, so no worries (: