Salandora/OctoPrint-FileManager #12: [Feature Request] Rename timelapse files

nbelakovski, 2018-01-02:
Would be great to extend this to be able to rename timelapses. Any idea if that's easy to do? If you point me in the right direction maybe I could implement it myself.

nbelakovski, 2018-01-03:
I've dug around a little bit and came up with a little hack, which didn't really work. I added a symbolic link to the timelapse folder within the uploads folder. I can see the timelapse folder now, but not the timelapses themselves. It seems the filemanager is filtering on .gcode and .stl. Or rather, it's using apis from octoprint which do that kind of filtering. I guess a new filetype would need to be added to get this to work, or maybe it would be easier to just add a rename button to the file list in the timelapse tab.

Salandora, 2018-03-04:
Add a button to the timelaps list.
That's much easier then trying to use a tool which is not designed to to that :-)

nbelakovski, 2018-03-05:
Not designed to do what?

Salandora, 2018-03-05:
The file manager plugin was not designed to rename timelapse files.
It wasn't even supposed to have something to do with the timelapse files.

nbelakovski, 2018-03-05:
Well, files are files right? It doesn't make a difference if they're gcode,
stl, or timelapse, they can still be deleted, renamed, copied, etc. It's
just a question of getting the list of files into the file manager, since
it seems there are some filters blocking the timelapse files from showing
up (as well as some path issues, but those are not as tough to solve)

Salandora, 2018-03-05:
Yeah, that's right, but while developing I never had in mind to list anything else then the gcode/stl files, so everything is tailored to these files and apis.
This is why I said it was never supposed to support the timelapse files.

It surely could be added somehow but at the moment I'm not planning to implement anything like this, cause there are other things to come first, like a progress bar, some bugs that have appeared.

Well on the other hand I can understand your point, and should add it to the todo list for a future version and maybe it would be a good idea to restructure the code, so new files can be added more easily...

For now I will reopen it.