Salandora/OctoPrint-FileManager #16: Nothing displayed

alexisju, 2018-06-05:
On a fresh install (Octopi/Octoprint 1.3.8 and FileManager 0.1.2), i had an issue where nothing (no one file) is displayed until i choose to set _Sort > Mixed_

If i turn back default setting _Sort > By Files / By Folders_ or _By Folders / By Files_, the listing goes empty again.

Henry0191, 2018-06-10:
Same problem here

christianlupus, 2018-06-22:
I can confirm this.

[image]

MichMich, 2018-06-24:
Can confirm.

Salandora, 2018-06-25:
Should be fixed with bug fix release 0.1.3
Please report back.

pauln, 2018-06-25:
0.1.3 does indeed seem to have fixed it for me - thanks for the release!

christianlupus, 2018-06-25:
Thanks, that seems to solved the problem.

alexisju, 2018-06-26:
Fixed :) Thank you!

closed 2018-06-26 by alexisju