Salandora/OctoPrint-FileManager #22: how do I get file-manager to show on the tab bar

mark-hahn, 2019-05-30:
I have put this in my ~/.octoprint/config.yaml ...
```
appearance:
  color: green
  components:
    order:
      tab:
      - temperature
      - control
      - gcodeviewer
      - terminal
      - file_manager
      - plugin_enclosure
  name: CR-10
```

... but the plugin_enclosure is showing before the file manager so it is pushed off to the drop-down which is a pita.

What am I doing wrong?

Is there some way to have all tabs shown via on of the many standard ways to do that?

Salandora, 2019-05-30:
I haven't testet it, because I'm lacking a proper system right now.

Because the fileManager is a plugin itself you should try to replace file_manager with plugin_filemanager
Please report back if that helped.

Also I don't know any way to show all tabs, this you should propably ask in the octoprint forum.

Vic-Y, 2020-06-12:
Hi. The following plugins may help:

OctoPrint-TabOrder allows users to re-arrange the order of the tabs, use icons on the tabs, and turn off the text labels. 
https://plugins.octoprint.org/plugins/taborder/

OctoPrint-WideScreen allows users to make use of more screen space on widescreen monitors.
https://plugins.octoprint.org/plugins/widescreen/

bitboy85, 2020-08-21:
I tried Taborder plugin and it works fine. So i guess this issue can be closed.

Salandora, 2021-01-24:
@bitboy85 confirmed it is working with TabOrder and therefore I'm going to close the issue.

closed 2021-01-24 by Salandora