Salandora/OctoPrint-FileManager #33: disableSelection error - plugin doesnt work

stefunkk, 2020-11-18:
Hi,

I have latest version of octoprint from docker, and file manager from plugin manager. When I switch window to file manager I cant do anything, web console shows me an error:

packed_plugins.js?c9da6618:716 Uncaught TypeError: $(...).disableSelection is not a function
    at HTMLDivElement.<anonymous> (packed_plugins.js?c9da6618:716)
    at HTMLDivElement.dispatch (packed_libs.js?ab739628:1)
    at HTMLDivElement.v.handle (packed_libs.js?ab739628:1)

I found also this:

https://api.jqueryui.com/disableSelection/#disableSelection

stefunkk, 2020-11-18:
https://github.com/Salandora/OctoPrint-FileManager/blob/master/octoprint_filemanager/static/js/ko.single_double_click.js

26 and 28 lines have to be removed.

Was it possible before to enter directory, because I can get into any.

Salandora, 2021-01-24:
@stefunkk sorry for the delay.
@jneilliii fixed the issue and made a PR. I just merged it.
Will make a compatibility release right away.

closed 2021-01-24 by Salandora