Salandora/OctoPrint-FileManager #35: No longer in development - Will you put plugin up for adoption?

evannadeau, 2020-12-13:
Per the guidelines from octoprint, I'm requesting that you put this plugin up for adoption as you have stated it is no longer in development.  Thank you for the great work you have done on it already.  I'm sure someone can continue your efforts and keep this great plugin going.

Thanks.

papperone, 2020-12-18:
it's a real shame this plugin is abandoned, there's even the patch to fix latest compatibility issue and nobody seems to take care to approve the pull requests :(

junebug217, 2021-01-18:
Please open this up for adoption.

Salandora, 2021-01-24:
Sorry to disappoint but no.
Reason: https://github.com/OctoPrint/plugins.octoprint.org/issues/719#issuecomment-766357378

closed 2021-01-24 by Salandora

evannadeau, 2021-01-24:
@Salandora I am not disappointed at all.  If you will be continuing support or working with Foosel to integrate into Octoprint as mentioned on the other thread [https://github.com/OctoPrint/plugins.octoprint.org/issues/719](url) I think all of the community can benefit.  Thanks for the great plugin.