Salandora/OctoPrint-FileManager #37: Cut and Paste Leaves Empty File in Original Location

cr45hmurphy, 2021-02-03:
Cut and Paste function works as expected. However, 130.0 byte file, with same name, is left in original location. See screenshots.

[image]
[image]

cr45hmurphy, 2021-02-04:
Going to close this issue. Appears that this is an issue with the [GCodeEditor](https://github.com/ieatacid/OctoPrint-GcodeEditor) plugin.

cr45hmurphy, 2021-02-04:
Can we reopen this issue, please? I thought I had isolated the issue to the GCodeEditor plugin, but that does not seem to be the case. It may be an issue with both of these plugins.

cr45hmurphy, 2021-02-04:
I thought I had isolated the issue to the GCodeEditor plugin, but that does not seem to be the case. It may be an issue with both of these plugins.

OllisGit, 2021-02-10:
Just for the records: Should be fixed in latest DLP release 1.25.4

closed 2021-02-11 by cr45hmurphy