Salandora/OctoPrint-FileManager #38: Star icon displaying incorrectly

cr45hmurphy, 2021-02-28:
First off, let me say that I love the new update! 0.1.6 is fantastic!

I am noticing that the star icons are getting scrunched up underneath the filename. This happens with/without the checkboxes enabled.

[image]

Salandora, 2021-03-03:
Uhm I never saw the Star icon, is that some kind of plugin?

ThorstenWirth, 2021-03-03:
https://github.com/eyal0/OctoPrint-PrintTimeGenius

cr45hmurphy, 2021-03-03:
> 
> 
> https://github.com/eyal0/OctoPrint-PrintTimeGenius

Are you saying that this plugin is causing the stars? If so, can you please explain how you have determined this. Clarification to your comment will help me better troubleshoot this issue on my instance.

cr45hmurphy, 2021-03-03:
> 
> 
> The star is just an indicator, the file was processed by the PrintTimeGenius plugin.

Thank you for this @ThorstenWirth. I found the setting within the plugin and was able to turn it off.

Salandora, 2021-03-06:
Well at this point in time I would like the issue to stay open as I think that either the star shouldn't appear in the first place or not be bugged out like this. So I'm looking into it and try to figure out if I can fix it on my side.

Salandora, 2021-03-11:
Fixed in Version 0.1.7rc1 for now.